TheDIYGuy999/Rc_Engine_Sound_ESP32 #2: cannot compile with Arduino IDE

lubopetrov, 2019-12-18:
When compiling for ESP32
ISR (SPI_STC_vect) is showing as error with description
expected constructor, destructor, or type conversion before '(' token

TheDIYGuy999, 2019-12-18:
1. Install ESP32 boards
2. Select the correct board
3. Never select / uncomment more than one sound in the same category
4. Install the statusLED library (you need the update from yesterday) https://github.com/TheDIYGuy999/statusLED

pstorti, 2019-12-24:
**Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Pedro\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Pedro\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Pedro\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\Pedro\AppData\Local\Temp\arduino_build_857616 -warnings=none -build-cache C:\Users\Pedro\AppData\Local\Temp\arduino_cache_26355 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\Pedro\Desktop\ENGINE\Rc_Engine_Sound_ESP32-master\Rc_Engine_Sound_ESP32\Rc_Engine_Sound_ESP32.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Pedro\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Pedro\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Pedro\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\Pedro\AppData\Local\Temp\arduino_build_857616 -warnings=none -build-cache C:\Users\Pedro\AppData\Local\Temp\arduino_cache_26355 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Pedro\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\Pedro\Desktop\ENGINE\Rc_Engine_Sound_ESP32-master\Rc_Engine_Sound_ESP32\Rc_Engine_Sound_ESP32.ino
Using board 'esp32' from platform in folder: C:\Users\Pedro\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\Pedro\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"C:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\Pedro\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/con [... cut ...]
Rc_Engine_Sound_ESP32:153:28: error: 'startSampleCount' was not declared in this scope

       if (curStartSample < startSampleCount) {

                            ^

Rc_Engine_Sound_ESP32:154:18: error: 'startSamples' was not declared in this scope

         a = (int)startSamples[curStartSample] + 128;

                  ^

Using library statusLED-master in folder: C:\Users\Pedro\Documents\Arduino\libraries\statusLED-master (legacy)
exit status 1
redefinition of 'const unsigned int sampleRate'**

TheDIYGuy999, 2019-12-24:
This issue should now be solved. There was an inconsistency in the uploaded sound files... Sorry, guys! Please re-download the entire package. It should then compile just fine.

pstorti, 2019-12-24:
confirming that it does compile and upload. thanks

TheDIYGuy999, 2019-12-24:
You're welcome

closed 2019-12-24 by TheDIYGuy999

pstorti, 2019-12-24:
but somehow i messed up the electronics and it does nothing lol