TheDIYGuy999/Rc_Engine_Sound_ESP32 #3: converting sound files

pstorti, 2020-01-01:
im using the website to convert sound files, and all i get is static

care to help?

btw, why is there only one rev sound? where is it called upon?

TheDIYGuy999, 2020-01-01:
Are the included files working? If not, something is wrong with your wiring.

The rev sound is not yet implemented and for testing only.

pstorti, 2020-01-01:
Included files work just fine, even the huge 007 music, with great quality.
Only the ones I made are sounding bad

Followed the instructions to the letter

TheDIYGuy999, 2020-01-01:
You need a lot of practice to prepare good sounding idle files in Audacity. Important:
- only cut the files on the exact zero crossing points
- try to find the "repeating" pattern (you can "pre-hear" the marked area with shift & tab)

closed 2020-01-13 by TheDIYGuy999