TheDIYGuy999/Rc_Engine_Sound_ESP32 #4: 2-speed manual gearbox support

AlfredBroda, 2020-03-06:
Adding 2-speed transmission support would be nice. In crawlers this would probably be a manual transmission (not throttle dependent). Having sounds and throttle managed while shifting would be very nice!

TheDIYGuy999, 2020-03-07:
a 3 speed manual is already supported, incl. sound & throttle management during shifting.

You jut have to change the shifting points (in ms) in gearboxDetection() to match your shifting servo:

// Gear detection
  if (pulseWidth[1] > 1700) selectedGear = 3;
  else if (pulseWidth[1] < 1300) selectedGear = 1;
  else selectedGear = 2;

AlfredBroda, 2020-03-07:
Ok, I'll look into it. Sounds like it only needs a different change channel to listen on. Does the throttle automatically ease off when `selectedGear` changes?

TheDIYGuy999, 2020-03-07:
The throttle is easing off, if you change to a higher gear and an additional throttle impulse is applied, if you change to a lower gear. This helps a lot to protect the shifting mechanism and sounds great. Have fun testing it!

closed 2020-03-20 by TheDIYGuy999