TheDIYGuy999/Rc_Engine_Sound_ESP32 #5: ideas: wastegate trigger edit,

Wombii, 2020-03-12:
While working with my automatic gearbox mod, I started reformatting the throttle calculations (mapThrottle() and engineMassSimulation() ) to make it easier to disable or modify parts of that code. These reformatted sections may or may not be useful to someone else, so I wanted to share it. 
[throttleCalculations.txt](https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/files/4322175/throttleCalculations.txt)

By adding the contents of throttleCalculations.txt to a new tab and doing the following edit in the main tab, everything should work as it does in default version 3.4.

Replace:
```
    // Map pulsewidth to throttle
    mapThrottle();
    // Simulate engine mass, generate RPM signal
    engineMassSimulation();
```
with:
 ```
   //Throttle calculations (map, volume and mass)
   throttleCalculations2();
```

Note that turboOverPressureTrigger2(); contains an edit to more reliably trigger the wastegate sound and volumeCalculation2() contains an edit to change the diesel knock volume by both rpm and load.

Wombii, 2020-03-13:
I made a mistake in the diesel knock volume calculation in that file. Here's a working one with volume based on a combination of rpm and load, for the default 3.4 scania setup:
```
// Calculate throttle dependent Diesel knock volume
    //Temp variables are necessary to prevent interrupt from using unconstrained values
  int tempthrottleDependentKnockVolume;
  int throttleDependentKnockVolume1 = dieselKnockIdleVolumePercentage;
  int throttleDependentKnockVolume2 = dieselKnockIdleVolumePercentage;
  int throttleDependentKnockVolume3 = dieselKnockIdleVolumePercentage;
  
  if (!escIsBraking && engineRunning) 
  {
    //First calculate knock based on rpm. More knock at higher rpms.
    tempthrottleDependentKnockVolume = map(currentRpm, 250, 400, dieselKnockIdleVolumePercentage, 110);
    throttleDependentKnockVolume1 = constrain(tempthrottleDependentKnockVolume,dieselKnockIdleVolumePercentage, 110);

    //Lower knock volume based on throttle input to reduce knock when dropping throttle.
    tempthrottleDependentKnockVolume = map(currentThrottle, 0, 450, dieselKnockIdleVolumePercentage, 70);
    throttleDependentKnockVolume2 = constrain(tempthrottleDependentKnockVolume,dieselKnockIdleVolumePercentage, 100);

    //Add extra knock based on difference between throttle and speed to simulate more knock when accelerating harder.
    tempthrottleDependentKnockVolume = map(currentThrottle-currentSpeed, 0, 500, 100, 500);
    throttleDependentKnockVolume3 = constrain(tempthrottleDependentKnockVolume,100, 500);
    
    //Calculate combined knock volume. Multiplying the values ensures knock volume is cut when releasing throttle.
    tempthrottleDependentKnockVolume = throttleDependentKnockVolume1 * throttleDependentKnockVolume3 / 100 * throttleDependentKnockVolume2/100;
    throttleDependentKnockVolume = constrain(tempthrottleDependentKnockVolume,dieselKnockIdleVolumePercentage, 100);
    
  }
  else throttleDependentKnockVolume = dieselKnockIdleVolumePercentage;

```

TheDIYGuy999, 2020-03-14:
Thank you for your feedback. I may include it in future versions

closed 2020-07-07 by TheDIYGuy999