TheDIYGuy999/Rc_Engine_Sound_ESP32 #7: statusLED object trouble

pstorti, 2020-03-20:
Tried with many boards, same result. 

Arduino: 1.8.12 (Windows 10), Board: "WEMOS LOLIN32, 80MHz, Default, 240MHz (WiFi/BT), 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Pedro\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Pedro\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Pedro\Documents\Arduino\libraries -fqbn=esp32:esp32:lolin32:FlashFreq=80,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\Pedro\AppData\Local\Temp\arduino_build_865706 -warnings=default -build-cache C:\Users\Pedro\AppData\Local\Temp\arduino_cache_212133 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Pedro\Documents\GitHub\Rc_Engine_Sound_ESP32\Rc_Engine_Sound_ESP32.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Pedro\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Pedro\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Pedro\Documents\Arduino\libraries -fqbn=esp32:esp32:lolin32:FlashFreq=80,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\Pedro\AppData\Local\Temp\arduino_build_865706 -warnings=default -build-cache C:\Users\Pedro\AppData\Local\Temp\arduino_cache_212133 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Pedro\Documents\GitHub\Rc_Engine_Sound_ESP32\Rc_Engine_Sound_ESP32.ino
Using board 'lolin32' from platform in folder: C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\Pedro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"C:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-IC:\\Users\\Pedro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protocomm" "-IC:\\Use [... cut ...]
                           ^

Rc_Engine_Sound_ESP32:1338:25: error: could not convert 'indicatorL.statusLED::flash(375ul, 375ul, 0ul, 0)' from 'void' to 'bool'

     if (indicatorL.flash(375, 375, 0, 0)) indicatorSoundOn = true;

                         ^

Using library statusLED-master in folder: C:\Users\Pedro\Documents\Arduino\libraries\statusLED-master (legacy)
exit status 1
no matching function for call to 'statusLED::begin(int, int, int, int)'

------------------------------------------------

AlfredBroda, 2020-03-20:
I had the same issue on a LOLIN32, but I only had that one type of board. Maybe we could have some control over which features are turned on or off? I don't actually need any lights for my Ural right now.

pstorti, 2020-03-20:
I have tried it in 3 different esp32 boards.

turning on or off stuff would be amazing. right now all i need is motor, horn, turning and break lights.

but first i need to actually compile it lol

AlfredBroda, 2020-03-20:
If turning features on/off would be done by `#ifdef FEATURE` it would help with compiling.

Wombii, 2020-03-20:
You have a Try renaming your statusLED-master folder in Documents\Arduino\libraries\ to just statusLED.  
"-master" is added to the zip file when downloading from github, but needs to be removed as the library name is "statusLED". 

I would also support adding an ifdef to disable leds and its pin assignments though (because I'm using a board with less i/o pins), but it only takes a few minutes to go through it myself, and different users would need to disable different things, so it's not straight forward to add that. At the moment ESC out also uses the statusLED library, so it's probably not done in an instant.

TheDIYGuy999, 2020-03-20:
Install the statusLed library and it will compile just fine. As Wombii says, different people want to disable different things, so it does not make sense to disable the LED code. If you don't want to use it, just leave it as it is and it will operate in the background. In general, this project is meant to be a base for your own code, which is adapted by you to fit your needs. I always try to add useful comments to the code, so it should not be too hard to program your own adaptations.

pstorti, 2020-03-20:
Library installed, chsnged the folder name and end up getting the same r3sult.

Seems like its confusing the statusled class with its subclasses, as it c9mplains about the numbwr of parameters

TheDIYGuy999, 2020-03-20:
You are using the newest statusLED library version? In order to make sure I uploaded everything correctly to GitHub, I just downloaded this project and the library. It's compiling just fine, even with -master in the library folder name. I'm using Arduino 1.8.10 on macOS, BTW. Are you sure, You installed this  ESP32 library in "Additional Boards"? https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
Maybe you have installed a 2nd copy of the statusLED library in a different folder?

TheDIYGuy999, 2020-03-20:
this tells me, you have installed an old version of the statusLED version:
C:\Users\Pedro\Documents\Arduino\libraries\statusLED-master/statusLED.h:21:10: note: candidate expects 3 arguments, 4 provided
Why? Because I added more arguments recently. You have to use the newest version

pstorti, 2020-03-21:
sorry about that. i had an old zip in my downloads folder and didnt realize the new has a [1] in it

one more thing. if I compile the december version, i can read signals from an electric bike throttle or a regular potentiometer... now, no matter if i comment the serial commands, the ppm, or both, it doesnt read anything. I have placed some Serial prints to monitor the changes, and it always reads 0

something specific to do  in order for it to read the analog signals?

thanks!

TheDIYGuy999, 2020-03-21:
No problem.

Which version do you mean? Please provide the version number. If I remember correctly, there was never potentiometer support in the ESP32 version.
If you want to use a potentiometer, use the analogRead() function

pstorti, 2020-03-21:
Ill check the versi9n when i get h0me.

I am familiar with analogread... used it in other projects but where do i deliver the results so it uses your code,? In the rc signal read section,?

TheDIYGuy999, 2020-03-21:
you need to modify this function:
void readRcSignals() {
  // measure RC signal pulsewidth:

  // CH1 Steering
  if (indicators) pulseWidth[0] = pulseIn(SERVO1_PIN, HIGH, 50000);
  else pulseWidth[0] = 1500;

  // CH2 (not used, gearbox servo)
  pulseWidth[1] = pulseIn(SERVO2_PIN, HIGH, 50000);

  // CH3 Throttle
  pulseWidth[2] = pulseIn(SERVO3_PIN, HIGH, 50000);
  if (pulseWidth[2] == 0) failSafe = true; // 0, if timeout (signal loss)
  else failSafe = false;

  // CH4 Additional sound trigger (RC signal with 3 positions)
  if (pwmSoundTrigger) pulseWidth[3] = pulseIn(SERVO4_PIN, HIGH, 50000);
  else pulseWidth[3] = 1500;

  // Invert RC signals
  invertRcSignals();

  // Falisafe for RC signals
  failsafeRcSignals();
}

Keep in mind, that you need to scale it to 1000 - 2000 milliseconds (= -45 to 45 RC servo degrees), before you store it in the pulseWidth[2] variable

closed 2020-03-22 by TheDIYGuy999