TheDIYGuy999/Rc_Engine_Sound_ESP32 #10: Suggestions/Doubts

praji181, 2020-05-25:
hi there, first of all let me thank you for this wonderful module you have created.  i use a GT PRO module and it is so limited. your work is awesome. Today successfully, i was able to connect everything to my truck and it worked like a charm, the sounds, the lighting and the dynamics. but there are some hiccups i came across, thought would let you know.

Just a few suggestions/doubts

1. I use a 6 channel 2.4Ghz box type transmitter. Can the channels or the stick be configured individually according to our preferences. ??
2.  Is there a way, to make the truck start at first using a channel( like GT PRO) and then it starts to roll when again an input on the stick is applied or stop it first, stays in idle and i get to turn the engine off manually.
3. I have an issue, when the forward channel is pressed the truck jumps first and then starts the engine. tried with the trimmers but no use.
4. can the lights and the horn be assigned switches, where we can turn it on and off, when we want. like turning on the hazard even though the truck is not on start mode, or leaving the headlights on even though the engine is not on. ( just an option)
5. turning on the engine and leaving the truck in idle and turning it off when we want is really a pleasurable thing to do. can you add that option as well.

But im not good with reading or interpreting the codes, maybe all these suggestions are already there in your code and i did not understand them or see them.

TheDIYGuy999, 2020-09-20:
Is your ESC connected to the ESC connector? Also note, that I don't recommend any other ESC than the Hobbywing 1080. I will work on the other topics, but it will take time. I have now ordered a Flysky stick based transmitter, so i can test this common setup.

TheDIYGuy999, 2020-10-08:
All your wishes are now implemented in v5.5

closed 2020-10-08 by TheDIYGuy999