TheDIYGuy999/Rc_Engine_Sound_ESP32 #12: General Questions

WarPorpoise, 2020-07-07:
Questions:
1. I noticed that some of the resistors in the pictures are blue indicating that they are probably metal-oxide resistors rather than the "typical" beige (carbon film) resistors. Can I use carbon film resistors, or are metal oxide resistors required?

2. Are 1/4 watt resistors sufficient for all "R" or should I use 1/2 watt (or more)?

3. Will BC337 NPN Transistors rated for 800ma & 45 Volt work for Q1-Q13?

4. Are the L, GND, R and +/- 5v connections on the PAM8403 soldered directly to the the connections on the board with some sort of jumper wire or are they soldered on pin connectors that are directly soldered to the board?

5. Does this appear to be the correct ESP32 dev kit for the newest 30 pin layout board? 
[Amazon ASIN B07Q576VWZ](https://www.amazon.com/MELIFE-Development-Dual-Mode-Microcontroller-Integrated/dp/B07Q576VWZ/ref=sr_1_1?dchild=1&keywords=B07Q576VWZ&qid=1594154193&sr=8-1)
I have never worked with an ESP32, and I am having an extremely hard time telling which versions is correct.

6. I noticed in the YouTube videos that you had a different set of speakers running on your rigs vs. the linked speakers on YouTube and was wondering if you purchased these or built you own? They sound very accurate and seem to have good bass response for their size.

JPS952, 2020-07-08:
Warporpoise, I used 1/4 watt resistors, in fact half watts are too big to fit well on the board. I used the same ESP32's as in your amazon link. As for the Pam8403, just the  pins on the one side of the board are soldered through using header pins. The outputs to the speakers do not connect to the host PCB. I also used a big heavy speaker from Amazon in my project:  https://www.amazon.com/gp/product/B000230LBI/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
this is an economical set of two speakers. I used one in my tank, check out JPS99 on Youtube.

TheDIYGuy999, 2020-07-08:
1. It does not matter
2. Any 1/4 Watt resistor will work
3. OK
4. Use 2.54mm male pin headers, directly soldered
5. OK
6. They are from this kit: https://www.banggood.com/Original-G_T_POWER-Lighting-And-Voice-Vibration-System-For-4WD-Tamiya-Tractor-Truck-RC-Car-Parts-p-1296706.html?utm_campaign=4828314_1046285&utm_content=1097&p=NZ29024828314201605E&cur_warehouse=CN

pavlosweb, 2020-07-27:
Did you try to add integrated wifi support or or bluetooth LE control? Perhaps web interface to make some settings?
I am very gratefull for your work. I am trying to build my own RC based on your code. It works like a charm. But when I try to use wifi or BLE it crashes. It has to do something with HW timers. When disabled it works well but I do not have suffient knowledge why is that or how to fix it. It seems you are in pretty good knownlege about esp32, do you have any idea?

TheDIYGuy999, 2020-07-27:
Yes, I tried to add WiFi upload, but it never worked properly. I think, enabling WiFi just takes too much resources. It's not on my priority list at the moment. I prefer to invest time in sounds. I'm also working on improving the additional functions triggering, using my new rcTrigger library.

pavlosweb, 2020-07-27:
> Yes, I tried to add WiFi upload, but it never worked properly. I think, enabling WiFi just takes too much resources. It's not on my priority list at the moment. I prefer to invest time in sounds. I'm also working on improving the additional functions triggering, using my new rcTrigger library.

I found similar issue https://www.reddit.com/r/esp32/comments/an8z16/how_to_move_timerbegin_interrupts_to_the_other/ It seems it is possible somehow. Dou you think I could get rid of hw timers and do its code in loop with appropriate delay or it is complete nonsense? 

I understand it is not your priority. Thank again for awesome work.

TheDIYGuy999, 2020-07-27:
I'm already using both cores. The timers are required for led control as well as for mills() and micros().

delay() is always nonsense in my opinion and probably the most stupid thing in there Arduino world.

closed 2020-08-26 by TheDIYGuy999