TheDIYGuy999/Rc_Engine_Sound_ESP32 #13: Tankmotorsound-2.h doesn't work properly with RC_Engine_Sound_ESP32.ino V5.1

JPS952, 2020-09-17:
I was behind in software revs  and went from Version 4.6 to 5.1.  I loaded the Actrox3363 settings as a baseline and uploaded them and they seemed to work okay.  I then changed the start sound to  tankstart3.h and that worked okay too.  I then added my Idle sound "tankmotorsound-2.h" and uploaded it and testing gave weird results.  While idling there is a loud repetitive1 hz knocking sound and the idle sound repeatedly revs up at about a 4hz rate.  I have gone back to the 4.6 version of software for now as it does what I need. If you would like to hear the sound let me know. jsangemino@yahoo.com

TheDIYGuy999, 2020-09-18:
If you make new sounds in Audacity:
- search for a cyclic pattern in the idle sound (the amount of ignition pulses is usually the same as the cylinder number), cut the "idle" sample to exactly this length, have a close look at the zero crossings to avoid clicking noises. The loudest peak should always be at the end of the sample.
- do the same with the "rev" sound. It will be 2 - 4 times shorter than the "idle" sample, depending on the engine and rpm of the "rev" sample
- Change the "Rate" (dropdown on the left of the sample) of the "rev" sample, until the length is the same as in the "idle" sample. This is very important!
- copy a part of the "rev" sample (the one with the original, unchanged "Rate" speed). This is the "knock" sample. Cut it to this max length: "Idle" length / number of cylinders / rpm range "MAX_RPM_PERCENTAGE" (usually 2 - 4 or 200 - 400%)
- Knock sound settings:
-- "dieselKnockInterval" = number of cylinders
-- uncomment "V8" for V8 engines or "V2" for V2 (Harley) engines
-- adjust "dieselKnockAdaptiveVolumePercentage" (how loud the "silent" knock pulses are compared with the loud ones), only active, if selected V8 or V2
- play with the other volumes, start-, end- and switch-points until you are happy
- The "rev" sound is optional and only active, if "REV_SOUND" is defined (// removed)
- Adjust the transition from the "idle" to the "rev" sound, using "revSwitchPoint", "idleEndPoint", "idleVolumeProportionPercentage". This step is very important and can make a huge difference!

I will add these informations to the v5.2 documentation. v5.2 will also include an optimised, pre made IS-3 tank sound preset

JPS952, 2020-09-18:
Thanks very much for the quick and detailed response and the updated version of software. I will apply the changes today. I appreciate how you have continuously updated and improved your software on all your projects!

closed 2020-09-18 by JPS952