TheDIYGuy999/Rc_Engine_Sound_ESP32 #24: Sound issues across different gearboxes

Whitey092, 2021-06-27:
Hi Martin,

Thanks for all your hard work, Love this project!

I've just updated to this latest version using the mercedes actross engine file. Normally I use the simulated 3 speed transmission, however it seems the engine does not 'rev up' to its limit. It sounds more dull or like its stuck in high gear. I makes the gear change noise when I actuate the switch but doesn't seem to change.

The 16 speed makes the correct engine noise, but I prefer the 3 speed.

Any chance you know what the fix would be?

Regards,
Sam

Whitey092, 2021-08-06:
So it appears it’s just the virtual 3 speed gearbox. It won’t rev up like the others do. How do we fix this?

TheDIYGuy999, 2021-11-07:
It should be fixed now

closed 2021-11-07 by TheDIYGuy999