TheDIYGuy999/Rc_Engine_Sound_ESP32 #27: Questions about PWM input

ws786873, 2021-09-25:
I used a 6ch PWM receiver and only connected CH3. In the software, I selected a remote controller recommended to use PWM communication settings, and annotated all types and models to use PWM. But I can't get the sound. What's wrong with me. When I use debug mode, it seems that channel detection cannot cycle. It always shows that channel 1 is out of the set range.

TheDIYGuy999, 2021-09-25:
Make sure, that "channelAutoZero" is set to "false" for all channels which are unconnected or don't have spring centering

ws786873, 2021-09-25:
I set all channels except 3CH to false in adjustmentremote, and my horn finally sounded, but it only increased output at 0-5% duty cycle and decreased output at 6% - 10%. 11% - 89% output idle sound, 90% - 100% is the same as 0-10%. I can see the throttle reading in the monitoring window. Most of the time, it is 0. So  is there any problem

TheDIYGuy999, 2021-09-25:
Looks like your throttle CH range is not 1000 to 2000 microseconds ( = -45 to 45° servo movement), which is standard. "Check your transmitter settings. You may want to lower "pulseSpan". What does "CHANNEL_DEBUG" say?

ws786873, 2021-09-25:
I can't understand what ch range is and why its unit is microseconds. In my understanding, it is a 0-3.3v square wave. I measured the output frequency of the receiver with a multimeter, which is 49hz, channel_ Debug say: the throttle value changes between 1099-2099. When my throttle is actually about - 40 ° - 40 °, channel_ Debug's throttle is displayed as 1599, which I think is a center position. In this way, the problem is too large a middle range. And I have a complete positive and negative range at the beginning and end.

TheDIYGuy999, 2021-09-25:
This is just how it was specified. Your throttle center position should be 1500. If not, "channelAutoZero" is not enabled or you touched the throttle during the initialization. Then it should work properly.

closed 2021-11-07 by TheDIYGuy999