TheDIYGuy999/Rc_Engine_Sound_ESP32 #28: Sound and flashing light

Frenazo, 2021-09-25:
Hi, 
The sound and light control has been working perfectly in tests with a simulated light board. I have installed everything in a Tamiya Actros and my surprise is that the sound plays normally if the lights are off, but if I turn the lights on the sound jumps when accelerating and the lights blink. 
The lights are fine if the engine is not revved. 
I use a 5,000mAm 7.2V NiMH battery, may I need a 3S Lipo (11.1V)? Thank you.

Regards

TheDIYGuy999, 2021-09-25:
Hi, I believe, that your 5V regulator is overloaded and cutting out, if the lights are on and the amplifier needs more current during revving. Are you using the original SMD board or what type is your regulator? Are you using series resistors for all LED? Maybe you want to post a picture of your setup?

Frenazo, 2021-09-25:
[image]

TheDIYGuy999, 2021-09-25:
Please answer the questions

Frenazo, 2021-09-25:
Sorry.
The board is JLCPCB (not SMD). 
I have mounted 2 SMD boards from JLCPCB and for some reason there is no communication between the ESC and the Receiver.

"Are you using series resistors for all LED?", Yes, sure, 1 resistor for each led

This is the provisional board and work perfectly:

https://www.youtube.com/watch?v=LIhg95roQEs

Frenazo, 2021-09-25:
The regulator is the BEC in Hobbywing 1080

TheDIYGuy999, 2021-09-25:
These could be potential issues:
- bad solder joint
- wiring error
- wrong series resistor
- 1000 uF capacitor soldered wrong way around

The regulator can't be the Hobbywing ESC BEC. Its voltage (6V) is too high for the amplifier and it will burn out. You need a 5V regulator on the board for the amplifier and for the LED.

Frenazo, 2021-09-25:
Ok.
The regulator in board is this: https://www.banggood.com/10pcs-L7805CV-TO220-L7805-TO-220-7805-LM7805-MC7805-Original-IC-p-1070097.html?utm_campaign=4828314_1046285&utm_content=1097&p=NZ29024828314201605E&cur_warehouse=CN

The board is the same in test and final truck, then I think "bad solder joint" is less probably.
Then I see "wiring error" or "wrong series resistor" first.

Thank, I'll try and we speak!

Frenazo, 2021-09-25:
Solved!
The problem was the potentiometer, I lowered the volume by turning the screw and it has been solved.
I sacrifice a few decibels to make everything work.
Thank you!!

Regards, Óscar

Frenazo, 2021-09-25:
[image]

closed 2021-09-26 by Frenazo