TheDIYGuy999/Rc_Engine_Sound_ESP32 #30: SMD Board Not Working

Whitey092, 2021-11-22:
Hi Martin, 
I can’t seem to get the smd boards working, I’ve had a batch made from 2 different suppliers including PCB way. I’ve checked all the pre soldered components and all are correct. 
When powered on the esp 32 lights up but I cannot get it to do anything else. However putting the same esp 32 board in the original PCB it runs without a problem. 
Am I missing something or is this a known issue?
Thanks for your help!

TheDIYGuy999, 2021-11-22:
Hi,
this not a known issue. I'm assuming there is a polarity issue with a diode or with an ULN2003. Did you check the voltage on the 5V rail, which is supplied by the on board 7805 regulator?

Whitey092, 2021-11-23:
Polarity all ok. 5V rail all good. I seem to have figured it out, SBUS needs to be inverted on smd boards. Does this sound right? Thanks for your help! Keep up the great work with this project! I love it!

closed 2021-11-23 by Whitey092