TheDIYGuy999/Rc_Engine_Sound_ESP32 #32: Cannot get it to compile on platform.io

sascha-schieferdecker, 2021-12-12:
Problem seem to be related to FASTLED and fastpin:

`.pio/libdeps/esp32dev/FastLED/src/platforms/esp/32/fastpin_esp32.h:136:125: error: template argument 3 is invalid
 #define _FL_DEFPIN(PIN) template <> class FastPin<PIN> : public _ESPPIN<PIN, ((uint32_t)1 << (PIN % 32)), _FL_PIN_VALID(PIN)> {};`

There are many more compiler errors like that. Is this a known problem?

sascha-schieferdecker, 2021-12-12:
Well, for anyone else stumbing across this problem, I could get a build if not using FASTLED 3.4.0, but 3.3.3.

I modified the platformio.ini like this:

`lib_deps =
    SPI
    https://github.com/TheDIYGuy999/statusLED
    https://github.com/TheDIYGuy999/SBUS
    https://github.com/TheDIYGuy999/rcTrigger
    https://github.com/bmellink/IBusBM
    https://github.com/Bodmer/TFT_eSPI
    https://github.com/FastLED/FastLED/archive/refs/tags/3.3.3.tar.gz`

sascha-schieferdecker, 2021-12-12:
Maybe the platformio.ino should have some version pinning in the lib_deps, so that only working versions are provided as dependency

TheDIYGuy999, 2021-12-15:
I added it in platformio.ini, but for me 3.4.0 is working as well

closed 2022-04-22 by TheDIYGuy999