TheDIYGuy999/Rc_Engine_Sound_ESP32 #33: Restart loop. Not a power issue

lolren, 2021-12-27:
My ESP32 loops at Initializing SBUS, check transmitter & receiver (sbusInverted true / false OK?)... then spits out some gibberish. nothing happens

TheDIYGuy999, 2021-12-28:
If you want to use SBUS, change sbusInverted true / false or vice versa. Also make sure, your receiver is powered up

lolren, 2021-12-28:
I did. Still nothing. But the problem is actually the restart of the ESP32. Sometimes it doesn't restart and has a different message, but the receiver is still not recognized, no mother what. Will try to order a new ESP32, but this one works with another projects.

TheDIYGuy999, 2021-12-29:
and you are using ESP32 board definition version 1.04 as mentioned in the sketch?
[image]
Otherwise it will crash.

lolren, 2021-12-31:
Yes. That`s the problem. Unfortunately, V1.04 is old enough an we cannot buy it, so the solution would be to see where the problem is and, if it`s not a vital function maybe disable it or change it. My board is 1.05 and I've ordered a 1.1 , and I`m still getting the same issue. The problem is that you cannot pick what board definition you buy, is kinda a lucky dip.
Ive notice that the restart loop goes away if I change the platform definition to a newer one (eg. espressif32@3.4.0) but then, ~I get another error, like ([E][WMath.cpp:74] map(): Invalid map input range, min == max), so the code must be changed and adapted to the newer boards definitions.

Regards

lolren, 2021-12-31:
I have tried to disable each library, one by one ( and the functions associated with them) in the code. Finally, when I disabled the watchdog  function the board stopped restarting. Now, I don`t know yet if the board is frozen and doesn't restart, or it will work, More investigation is needed. Regards

TheDIYGuy999, 2021-12-31:
v1.04 is not related to the hardware. It's the board definition, provided by espressif. You have to use 1.04:
[image]

lolren, 2022-01-04:
Hi. Sorry for the late reply, been busy with the new year😁 
I've installed v1.04 , but this didn't change anything. Still getting weird restarts after a few seconds, related to watchdog.
Now I think my remote control protocol is not supported 
But this is ok, as I'm building the unit for a friend of mine that uses a Flysky.

Thank you

lolren, 2022-01-05:
Managed to fix the restart loop issue, by disabling the watchdog. 
But, now with a FLYSKY_FS_I6X  , I cannot get the board to sync for some reason. CH 1 signal out of auto calibration range, check transmitter & receiver is the error I get, no matter what I choose from the remote option. Regards

TheDIYGuy999, 2022-01-05:
If you have a valid SBUS signal coming in, you don't have to disable the watchdog. Looks like your steering CH1 is not in center position:
[image]

lolren, 2022-01-15:
You can close the issue. The problem was, indeed the  ESP32 board definition above 1.04. 
Also the reason that the remote didn`t sync was that I had no idea the receiver must have  SBUS output. 
I was stupidly trying to connect to the PWM output. 
Thank you for the support. 
Regards

fiechr, 2022-02-19:
I assume, that I have the same issue (using PlatformIO)

The platform parameter in the platformio.ini is set to "platform = espressif32@3.0.0"

Some LEDs are blinking and the speaker makes a popping noise every few seconds. The serial monitor repeats the same message about S-BUS initialization endlessly every few seconds.

This is my first board, so in addition I also don't know what it is supposed to do.

I've tried multiple receivers, PCBs and ESP32 dev boards. I also tried inverted/non-inverted.

But it seems to have nothing to do with that.

lolren, 2022-02-19:
It's the receive. You need a speci ibus or sbus receiver. That solved my problem. Regards

fiechr, 2022-02-19:
I've tested a Flysky FGr4v2 (with NB4 radio), an A8S and a iA6B (with I6 radio), as well as a Radiolink R7FG.

All receivers were set to SBUS mode in the RX settings.

lolren, 2022-02-19:
But, did you connect the esp to the SBUS output, not a PWM output?

fiechr, 2022-02-19:
I'm pretty sure, yes. Some have dedicated SBUS output pins, others use ports like "SERVO" or "IBUS".

Is there any way to get more information on why it fails exactly? Is it possible to configure the radio or its configuration wrong, so it reproduces this kind of issue? I will check out the debug settings on the *.ino file.

I've tried different radios with my own settings and also the I6 with the stock I6X profile, but still, no SBUS initialization.

fiechr, 2022-02-19:
After trying different setups and hardware, I finally tried set `sbusInverted` again and now it works...
Maybe a combination of different settings.

The FrSky FGr4 and iA6B seem to use inverted SBUS signals.

closed 2022-04-22 by TheDIYGuy999