TheDIYGuy999/Rc_Engine_Sound_ESP32 #36: Use throttle range from 1000-2000 with no reverse gear

derFrickler, 2022-02-25:
Is it possible to configure the throttle range from 1000-2000 such as motor off is at 1000 and full throttle is at 2000
No reverse gear needed. 

This would be great for models (airplanes) that don't have reverse gears. 
Any hints on how to use this would be great. 
So far i am using PWM input with all channels but Throttle/3 set to NONE.

TheDIYGuy999, 2022-03-05:
Done in v9.0

closed 2022-03-05 by TheDIYGuy999

derFrickler, 2022-03-05:
Thanks! I got it working already with a dirty fix directly in the readPWM
`    
int offs = (pulseWidth[3]-1000)/2;
pulseWidth[3] = 1500+offs;
`
Will Update to 0.9 then.
BTW, is there a way to stop the engine via PWM signal? Maybe if going lower than 1000?