TheDIYGuy999/Rc_Engine_Sound_ESP32 #48: PCBWay component help

BusDriverLee, 2022-08-11:
I've tried to order the boards using the files section with PCBWay, but they have quoted the incorrect SMD-ECAP-10x10 for C1. They have quoted part number EEE-1AA102P which is too big. I believe it should be EEEFC1A102AP but this has a different stamp compared to your picture examples. Any help with the correct part number would be much appreciated.

fiechr, 2022-08-12:
Looking at the Digikey database these two seem to have the same size (dimensions and value). Both are 10x10 mm and 1000µF. Also the footprints look identical (same dimensions). Only different series and longer lifetime.

https://www.digikey.de/de/products/detail/panasonic-electronic-components/EEE-1AA102P/766035
https://www.digikey.de/de/products/detail/panasonic-electronic-components/EEE-FC1A102AP/1141901

Or am I missing something? What do you mean with "too big"?

BusDriverLee, 2022-08-12:
[FileFromPCBWay.xlsx](https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/files/9316714/FileFromPCBWay.xlsx)
Hi @fiechr, 
Thank you for your quick reply. Please see attached the spreadsheet that was sent to me by PCBWay. They are stating that the components they have will fall off the board. Hopefully this sheet will make sense to you as it doesn't to me! I wonder if they have been sent the wrong component?

fiechr, 2022-08-12:
I have to agree, that the pads are pretty short and I also had some difficulties when hand-soldering the capacitor on my own PCBs. 

On the other hand, the footprint should be big enough. I don't know why they now reject it, since, I assume, others have ordered the PCB including assembly in the past successfully. Other SMD components, like linear voltage regulators or transistors also use contact areas below the actual device. In this case the whole PCB needs to be heated up, because there is no other way to make the solder flow in between.

The caps I used (see image below) have the same size they show in the picture inside the spreadsheet file. It's pretty tight, but works.

[image]

Without changing the footprint on the board itself, I think, the only way is to either solder the capacitor later yourself by hand or replace it with a smaller one, so they accept it, even if it most likely will have a lower capacity, it should still be fine for this application.

BusDriverLee, 2022-08-23:
Hi @fiechr,

Thank you for your help. I managed to find an 8mm that seems to have the same specifications - EEE-FT1A102GP. I have sent this to them and they have assembled a board. I am awaiting delivery so I can try it out..... fingers crossed!

[image]

Lee