TheDIYGuy999/Rc_Engine_Sound_ESP32 #54: Compilation error while uploading

Draxuxe, 2023-01-24:
Hello,

When I use Arduino IDE to transfer your code to the esp32, there is a bunch of compilation error popping up : "Compilation error: redefinition of '...'" and i only uncommented one .h for my vehicles.
Thanks for the help

fet43, 2023-02-16:
good job I already copied the code but it always gives the same error was it possible to help ??

In file included from c:\Users\acer\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                  from C:\Users\acer\Desktop\files\sun\Rc_Engine_Sound_ESP32\src\src.ino:99:
c:\Users\acer\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: [#pragma](https://www.youtube.com/hashtag/pragma) message: No hardware SPI pins defined. All SPI access will default to bitbanged output
  # pragma message "No hardware SPI pins defined. All SPI access will default to bitbanged output"

TheDIYGuy999, 2023-03-01:
redefinition means you uncommented multiple options at the same time.

No hardware SPI pins defined. can be ignored

closed 2023-03-01 by TheDIYGuy999