TheDIYGuy999/Rc_Engine_Sound_ESP32 #57: sound issues

racin44, 2023-02-10:
This is my second soundboard of this and the only sound i get is clicking. The only difference is I'm using less of the setup, Lighting. This is for a rc race car. I been trying multiple esp32 boards to see if that's the issue, but its not.

TheDIYGuy999, 2023-02-13:
What does the serial monitor say?

racin44, 2023-02-13:
A2      : 0x00000000  A3      : 0x00000005  A4      : 0x3ffbdcf4  A5      : 0x00000000  
A6      : 0x00000005  A7      : 0x00000001  A8      : 0x3ff59064  A9      : 0x3ff59000  
A10     : 0x00000000  A11     : 0x3ffc3b88  A12     : 0x3ffc3b88  A13     : 0x00060623  
A14     : 0x00060620  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008501d  LEND    : 0x40085025  LCOUNT  : 0x00000027  

Backtrace: 0x400d61f6:0x3ffdd710 0x400d3e47:0x3ffdd740 0x400d3249:0x3ffdd760 0x400d2493:0x3ffdd780 0x400d2c9f:0x3ffdd7b0

ELF file SHA256: 36f1d3323bc41389

Rebooting...