TheDIYGuy999/Rc_Engine_Sound_ESP32 #59: Control issues volume dependant

Whitey092, 2023-04-05:
Hi Martin, 
I’m having an issue with control dependant on volume. 
There is no issue when at maximum and 1 notch down. 
At lowest and muted volume levels any speed/throttle/brake settings are ignored. The ESC accelerates as if it is not controlled by the board, will not brake and has no inertia.

I made a new board with a new esp32 just to rule that out but it didn’t make a difference.

FBester1976, 2023-04-05:
Hello,

The board is put in "Crawler mode" based on values in 8_adjustmentsSound.h. So, in crawler mode the virtual inertia is disabled.

If you do not want Crawler mode, you need to make the **masterVolumeCrawlerThreshold** less than the lowest Volume percentage you have. If you make the masterVolumeCrawlerThreshold = 43 or less, you will not have Crawler mode for the current selection below.
 
// Volume adjustment
const  uint8_t numberOfVolumeSteps = 3; // The mumber of volume steps below
const uint8_t masterVolumePercentage[] = {100, 66, **44**}; // loud, medium, silent (more than 100% may cause distortions)

//const  uint8_t numberOfVolumeSteps = 4; // The mumber of volume steps below
//const uint8_t masterVolumePercentage[] = {100, 66, **44**, 0}; // loud, medium, silent, no sound (more than 100% may cause distortions)

// Crawler mode
const uint8_t **masterVolumeCrawlerThreshold** = **44**; // If master volume is <= this threshold, crawler mode (without virtual inertia) is active

Hope this helps...

TheDIYGuy999, 2023-04-06:
Thank you, Frikkie!

closed 2023-04-06 by TheDIYGuy999

Whitey092, 2023-04-24:
Ahhh perfect! Thank you. 
I have one more question.
When using Virtual 3 Speed and wireless trailers, the truck engine sound goes to full rev as soon as you start the truck. If I move the throttle a little bit it comes back down. But with throttle at neutral it revs high. Am I missing something that controls that?

FBester1976, 2023-04-24:
Some basic questions:

What remote do you use?
Do you have a self-centring spring on the throttle?
Do you always switch the remote on before the truck?
Possible to make a video? Youtube link?

INTERNAL
Volkswagen of South Africa (Pty) Ltd. (Reg No. 1946/023458/07)

Chairperson & Managing Director:

M Biene*

Directors:
R Coleman (Human Resources)
G Dozo** (Finance)
N Maliza (Corporate & Government Affairs)
T Milz* (Sales and Marketing)
U Schwabe* (Production)

N Matyumza (Non- Executive)
M Raphulu (Non-Executive)

German*
Argentinian**

Volkswagen South Africa (Pty) Ltd is proudly Level 3 B-BBEE compliant

DISCLAIMER :  Volkswagen of South Africa (Pty) Ltd

Any views expressed in this message are those of the individual sender.
No liability shall attach whatsoever to Volkswagen Group SA from this communication except where the sender is acting on specific authority of Volkswagen Group SA, such authority being public record and acknowledged by Volkswagen Group SA by nature of the employee's functions. This document may in no way be photocopied, printed, scanned or
electronically duplicated for any purposes other than that for which it was originally intended.

If you are not the intended recipient of this communication, please discard this message and notify Volkswagen Group SA immediately at ***@***.*** ***@***.***>

Volkswagen Group SA's anonymous toll free ethics number is: 080 111 3949

Whitey092, 2023-04-28:
Hi Frikkie, 
Thanks for your response. I am using Flysky Paladin EV. Self centering sticks, and yes transmitter always powered on before reciever.

As I said this only happens with wireless trailers enabled. If I uncommonly wireless trailer it works perfectly. 

Here is a YouTube short link:
https://youtube.com/shorts/ncpT9OmvEzY?feature=share

I have tried it with trailers connected also but it does not fix the problem.