TheDIYGuy999/Rc_Engine_Sound_ESP32 #65: BOM.xls in gerbers.xls

ArchWinDoors, 2023-10-06:
I have followed the How to Order from PCBWay, received assembled board - Just wondering why many of the components are struck-out: e.g WAGO SREW CLAMP.
These components missing, means that I don't have everything I need to set up and finish the board. 
Basically the BOM included differs from what's shown at: https://youtu.be/csQgTfxRd8Y?t=102
I'm novice so I didn't know items were missing when PCBway sent the confirmation..
PCBWay have followed the files sent to them and I can't really raise a dispute if they've followed the BOM.xls sent from the repo.

FBester1976, 2023-10-06:
Hello,

If I understand you correctly, you are missing all the through hole components?

In the video and on the instruction (https://youtu.be/csQgTfxRd8Y?si=0IE6IQnMq7uUKS0f&t=75) they speak about SMD (Surface mount) and through hole components. If you ordered according to the video and the instructions, you would have specified no through hole components.

You need to order those components separately from a supplier in your area.
***@***.***
SMD assembly from PCBWay
***@***.***

After I soldered all the through hole components. (as the YouTube video explains)
***@***.***

Below are just some links so you can see what you need (not sure if you are also planning the wireless trailer card?)

PAM8403 5V Amp, x1 truck  sound card, none on trailer card.
https://www.amazon.in/ICSTORE-PAM8403-Amplifier-Speaker-Subwoofer/dp/B07HLB1HMR

5pcs MP1584 5V Fixed Output 3A DC Step Down Converter Buck Module 4.5-28V Input x2 (needed for truck and trailer)
https://www.jacobsparts.com/items/DCMOD-I-5V

RAMPS PCB CONNECTOR SCREW TYPE x2 (needed for truck and trailer)
https://www.diyelectronics.co.za/store/connectors-wiring/296-ramps-pcb-connector-screw-type.html?search_query=screw+connector&results=84

20K Ohm 203 Multi turn Trimpot Potentiometer / Variable Resistor, (x1 truck  sound card, none on trailer card.)
https://electrobes.com/product/20k-ohm-203-multi-turn-trimpot-potentiometer-variable-resistor/

SINGLE ROW 40-PIN 2.54MM PITCH PIN HEADER (RED, BLUE, YELLOW, WHITE OR BLACK)  I prefer coloured pins, makes plugging in easier.
https://hobbycomponents.com/connectors/741-single-row-40-pin-254mm-pitch-pin-header-red-blue-yellow-white-or-black

1×40-Pins 2.54mm SIL Female Header (if you can get 15-pins, you do not have to cut the 40-pin)
https://handsontec.com/index.php/product/40-pins-sil-female-headers/

ESP32 Dev board 30PIN x2 (needed for truck and trailer)
https://www.walmart.ca/en/ip/Loewten-ESP32-Development-Board-WiFi-ESP32-Development-Board-ESP32-Development-Board-Module-30-Pin-2-4GHz-WiFi-2-In-1-Microcontroller-Processor/PRD5GER7QKF52AF

Regards,

Frikkie

INTERNAL
Volkswagen of South Africa (Pty) Ltd. (Reg No. 1946/023458/07)

Chairperson & Managing Director:

M Biene*

Directors:
R Coleman (Human Resources)
G Dozo** (Finance)
N Maliza (Corporate & Government Affairs)
T Milz* (Sales and Marketing)
U Schwabe* (Production)

N Matyumza (Non- Executive)
M Raphulu (Non-Executive)

German*
Argentinian**

Volkswagen South Africa (Pty) Ltd is proudly Level 3 B-BBEE compliant

DISCLAIMER :  Volkswagen of South Africa (Pty) Ltd

Any views expressed in this message are those of the individual sender.
No liability shall attach whatsoever to Volkswagen Group SA from this communication except where the sender is acting on specific authority of Volkswagen Group SA, such authority being public record and acknowledged by Volkswagen Group SA by nature of the employee's functions. This document may in no way be photocopied, printed, scanned or
electronically duplicated for any purposes other than that for which it was originally intended.

If you are not the intended recipient of this communication, please discard this message and notify Volkswagen Group SA immediately at ***@***.*** ***@***.***>

Volkswagen Group SA's anonymous toll free ethics number is: 080 111 3949

TheDIYGuy999, 2023-10-07:
Thank you, Frikkie

ArchWinDoors, 2023-10-09:
@FBester1976 Thank you, this makes sense now and I appreciate the links to source equivalent components.

closed 2023-10-09 by ArchWinDoors