TheDIYGuy999/Rc_Engine_Sound_ESP32 #67: Sound

ryadavoi, 2024-02-13:
Hello, I only need sound for my project on esp32 arduino ide, but I can’t get just the code for sound

rengels, 2024-02-14:
The header files containing the sounds are what you need then.

On my branch there is a tool to convert the header files back to wav files: https://github.com/rengels/Rc_Engine_Sound_ESP32/blob/huracan/tools/Header2Audio.html

If you want to use the sounds on an arduino you would need to convert them back again.

ryadavoi, 2024-02-18:
I don’t need sound files, i need library for sound playback , without motors, leds and others

rengels, 2024-02-18:
Then this is not your project.
It uses way to much HW if you just want to play sound.

I googled something for you:
https://www.arduino.cc/reference/en/libraries/soundplayer/