TheDIYGuy999/Rc_Engine_Sound_ESP32 #68: Engine sound controlled by Analog input horn controlled by digital input

vinhx, 2024-02-26:
Hello sir,
your project so great!
I'm currently making an 3 wheels electric bike for my children. that the motor speed controlled by a hall sensor , the signal output is analog 0 to 5v. My idea is use the analog input on esp32 to control engine sound, the horn controlled by digital input pin. Could you please give me advice how to get those relative code module from whole project
thank you alot for all your great work and kindly sharing!!!

Best regard!

rengels, 2024-02-26:
Thanks for the praise,

The SW isn't really that modular. 
If you are fit in programming, you can replace the pwm input functions by something else.

An other solution would be to convert your 0-5v signal and the horn switch to PWM signals via an external (small) arduino.
Here is something that I found:
https://forum.arduino.cc/t/creating-your-own-pwm-to-control-a-servo/129869

Good luck

vinhx, 2024-02-27:
oh, I got your point!
I will try to replace the PWM signal by data mapping from analog input, thank you alot!