TheDIYGuy999/Rc_Engine_Sound_ESP32 #69: ESP32 Blue Light Blinking !!Help Please!!

RS33D, 2024-03-15:
Hello

Your ESP32 engine sound project is very nice, currently trying to put it into my 3d printed 1/14 scale kenworth 

But i have an issue with the circuit board not working, please see the below google drive video

[https://drive.google.com/file/d/1ZZUraMMt_CvwXIOm6ZMI8HmvmG5yAJjD/view?usp=sharing](url)

I have not adjusted the code as i want to use Sbus so just loaded the code up, but as per the video you can see that when turned on the blue light just flashes 

i have adjusted the settings on the remote i am using which is a Flysky FS i6x and has its serial output set on S Bus, the receiver is also s bus compatible, i am using a flysky ia10b s receiver

i have also tried ibus, PWM and PPM and all result in the same issue  

What have i done wrong??

your help would be much appreciated 

many thanks 

Lawson

FBester1976, 2024-03-15:
Good morning,

Some questions…

What does the serial monitor say? Keep your ESP32 plugged into the PC when you start it…serial monitor will state errors.
***@***.***
What version of software are you using?
Do you have the battery protection activated (3_ESC tab)?
***@***.***

I can see from your video that your ESC is plugged into the wrong position (not Ch1, needs to go to ESC.)
***@***.*** ***@***.***

Also plug in a speaker when you make the next video please.

Regards,

Frikkie

INTERNAL

Volkswagen Group Africa 
Chairperson & Managing Director:

M Biene*

Directors:
R Coleman (Human Resources)
N Maliza (Corporate & Government Affairs)
T Milz* (Sales and Marketing)
F Ribeiro** (Finance)
U Schwabe* (Production)

M Raphulu (Non-Executive)

German*
Brazilian**

Volkswagen of South Africa (Pty) Ltd is proudly Level 3 B-BBEE compliant

DISCLAIMER :   Registered Name: Volkswagen of South Africa (Pty) Ltd  (Reg No. 1946/023458/07), trading as Volkswagen Group Africa

Any views expressed in this message are those of the individual sender. 
No liability shall attach whatsoever to Volkswagen of South Africa from this communication except where the sender is acting on specific authority of Volkswagen of South Africa, such authority being public record and acknowledged by Volkswagen of South Africa by nature of the employee's functions. This document may in no way be photocopied, printed, scanned or 
electronically duplicated for any purposes other than that for which it was originally intended. 
If you are not the intended recipient of this communication, please discard this message and notify Volkswagen of South Africa immediately at ***@***.***
Volkswagen of South Africa's anonymous toll free ethics number is: 080 111 3949

RS33D, 2024-03-15:
Okay, thanks for your reply

Here is what the serial monitor says 

**************************************************************************************************
TheDIYGuy999 RC engine sound & light controller for ESP32 software version 9.12.0
https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32
Please read carefully: https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md
XTAL Frequency: 40 MHz, CPU Clock: 240 MHz, APB Bus Clock: 80000000 Hz
Internal RAM size: 289660 Byte, Free: 264244 Byte
WiFi MAC address: E4:65:B8:83:E7:48
Core 0 reset reason: 12: SW_CPU_RESET
Core 1 reset reason: 12: SW_CPU_RESET
**************************************************************************************************

EEPROM read.
current eeprom_id: 2
change it for default value upload!

Warning, BATTERY_PROTECTION disabled! ESC with low discharge protection required!
-------------------------------------
ENABLE_WIRELESS option enabled

Informations for web configuration via your cell phone or computer *******************************
SSID: My_Truck
Password: 123456789
IP address: 192.168.4.1

WiFi Tx Power Level: 78
WiFi Tx Power Level changed to: 28

The following trailers are currently enabled:
Trailer 1 MAC address: FE:49:01:00:00:01
-------------------------------------

**************************************************************************************************

Regarding the software i am using 

Ardunio IDE is 1.8.15 
Diyguy999 code version i am using is 9.12.0

No, i have not removed the \\ from battery protection, so I am assuming it is not activated 

Please see the below video for when battery protection is enabled (ie \\ removed)

[https://drive.google.com/file/d/1QNNL5fuSMhIF74WCbjhMlbAaCGyKIKRQ/view?usp=sharing](url)

When battery protection is enabled, serial monitor will say this once, even if i am running it off my 2s lipo battery and just have the usb cord plugged into my computer  

**************************************************************************************************
TheDIYGuy999 RC engine sound & light controller for ESP32 software version 9.12.0
https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32
Please read carefully: https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md
XTAL Frequency: 40 MHz, CPU Clock: 240 MHz, APB Bus Clock: 80000000 Hz
Internal RAM size: 289272 Byte, Free: 263856 Byte
WiFi MAC address: E4:65:B8:83:E7:48
Core 0 reset reason: 1: POWERON_RESET
Core 1 reset reason: 14: EXT_CPU_RESET
Battery protection calibration data:
RESISTOR_TO_BATTTERY_PLUS: 10000 Ω
RESISTOR_TO_GND: 1000 Ω
DIODE_DROP: 0.00 V
**************************************************************************************************

EEPROM read.
current eeprom_id: 2
change it for default value upload!

Battery voltage: 1.58 V
Cutoff voltage per cell: 3.30 V
Fully charged voltage per cell: 4.20 V
Error, no valid battery detected! Only 2S & 3S batteries are supported!
REMOVE BATTERY, CONTROLLER IS LOCKED = 2 FAST FLASHES!

thanks for your comments so far.

FBester1976, 2024-03-16:
Good morning,

If you have not fitted the battery protection resistors on the back of the card, keep the // in place.

I also noticed that you have not powered the card through a y-lead from your battery.

[cid:25BAFFE4-087B-4324-B895-A9AD3ACF2061]

[cid:79CB54DC-D45A-427A-A690-8C273FBEAAEE]

I also do not see a speaker connected for the sound.

Regards,

Frikkie

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

RS33D, 2024-03-16:
Okay, should I be taking power from the battery to the esc as well as the
screw terminals on the circuit board?, is that what you mean by powering
through a y lead from the battery

FBester1976, 2024-03-16:
That is correct

RS33D, 2024-03-16:
Okay, please see the Google drive video, have just used jumper leads to
take power to circuit board, have also plugged in a speaker, which makes a
noise like a cricket when turned on

https://drive.google.com/file/d/1ZwvYzqQYg46y0y7oNw9ggOLlnbyNeDA-/view?usp=drivesdk

Many thanks

Lawson

FBester1976, 2024-03-16:
I think you still have the battery protection active

https://drive.google.com/file/d/1ZwvYzqQYg46y0y7oNw9ggOLlnbyNeDA-/view?usp=drivesdk

RS33D, 2024-03-16:
Have turned off battery protection, but still no luck, see the below video

https://drive.google.com/file/d/1__MXfzIHpZRCOB1SxroqZxn1pJdPsphG/view?usp=drivesdk

Many thanks

RS33D, 2024-03-16:
Sure.

 

**************************************************************************************************

TheDIYGuy999 RC engine sound & light controller for ESP32 software version 9.12.0

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

Please read carefully: https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md

XTAL Frequency: 40 MHz, CPU Clock: 240 MHz, APB Bus Clock: 80000000 Hz

Internal RAM size: 289660 Byte, Free: 264244 Byte

WiFi MAC address: E4:65:B8:83:E7:48

Core 0 reset reason: 8: TG1WDT_SYS_RESET

Core 1 reset reason: 14: EXT_CPU_RESET

**************************************************************************************************

 

EEPROM read.

current eeprom_id: 2

change it for default value upload!

 

Warning, BATTERY_PROTECTION disabled! ESC with low discharge protection required!

-------------------------------------

ENABLE_WIRELESS option enabled

 

Informations for web configuration via your cell phone or computer *******************************

SSID: My_Truck

Password: 123456789

IP address: 192.168.4.1

 

WiFi Tx Power Level: 78

WiFi Tx Power Level changed to: 26

 

The following trailers are currently enabled:

Trailer 1 MAC address: FE:49:01:00:00:01

-------------------------------------

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

 

Core  1 register dump:

PC      : 0x40082511  PS      : 0x00060435  A0      : 0x40085c60  A1      : 0x3ffbf70c  

A2      : 0x3ffda494  A3      : 0x3ffbdd7c  A4      : 0x00000000  A5      : 0x00000000  

A6      : 0x000000b5  A7      : 0x00060d23  A8      : 0x3ff60000  A9      : 0x00000001  

A10     : 0x00000000  A11     : 0x3ffc5e40  A12     : 0x3ffc5e44  A13     : 0x00060423  

A14     : 0x00060421  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x00000006  

EXCVADDR: 0x00000000  LBEG    : 0x4008a5e8  LEND    : 0x4008a5f3  LCOUNT  : 0xffffffff  

Core  1 was running in ISR context:

EPC1    : 0x400ebb0f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

 

 

Backtrace: 0x4008250e:0x3ffbf70c |<-CORRUPTED

 

 

 

 

ELF file SHA256: 360c5eb004a87de0

 

Rebooting...

 

**************************************************************************************************

 

 

***@***.***>

FBester1976, 2024-03-16:
Is the trim on your steering channel on zero?

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

RS33D, 2024-03-16:
yes

 

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

RS33D, 2024-03-16:
I have noted that when viewing the serial monitor with battery protection
off it sends the same information over and over again, however with battery
protection on, it only sends information once.

Is it meant to do this?

FBester1976, 2024-03-16:
Yes, what is the info it is sending?

RS33D, 2024-03-16:
Says this over and over again, should I try uploading the code through VS
studio

**************************************************************************************************

TheDIYGuy999 RC engine sound & light controller for ESP32 software version
9.12.0

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

Please read carefully:
https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md

XTAL Frequency: 40 MHz, CPU Clock: 240 MHz, APB Bus Clock: 80000000 Hz

Internal RAM size: 289660 Byte, Free: 264244 Byte

WiFi MAC address: E4:65:B8:83:E7:48

Core 0 reset reason: 8: TG1WDT_SYS_RESET

Core 1 reset reason: 14: EXT_CPU_RESET

**************************************************************************************************

EEPROM read.

current eeprom_id: 2

change it for default value upload!

Warning, BATTERY_PROTECTION disabled! ESC with low discharge protection
required!

-------------------------------------

ENABLE_WIRELESS option enabled

Informations for web configuration via your cell phone or computer
*******************************

SSID: My_Truck

Password: 123456789

IP address: 192.168.4.1

WiFi Tx Power Level: 78

WiFi Tx Power Level changed to: 26

The following trailers are currently enabled:

Trailer 1 MAC address: FE:49:01:00:00:01

-------------------------------------

Guru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1).

Core 1 register dump:

PC : 0x40082511 PS : 0x00060435 A0 : 0x40085c60 A1 : 0x3ffbf70c

A2 : 0x3ffda494 A3 : 0x3ffbdd7c A4 : 0x00000000 A5 : 0x00000000

A6 : 0x000000b5 A7 : 0x00060d23 A8 : 0x3ff60000 A9 : 0x00000001

A10 : 0x00000000 A11 : 0x3ffc5e40 A12 : 0x3ffc5e44 A13 : 0x00060423

A14 : 0x00060421 A15 : 0x00000001 SAR : 0x00000020 EXCCAUSE: 0x00000006

EXCVADDR: 0x00000000 LBEG : 0x4008a5e8 LEND : 0x4008a5f3 LCOUNT :
0xffffffff

Core 1 was running in ISR context:

EPC1 : 0x400ebb0f EPC2 : 0x00000000 EPC3 : 0x00000000 EPC4 : 0x00000000

Backtrace: 0x4008250e:0x3ffbf70c |<-CORRUPTED

ELF file SHA256: 360c5eb004a87de0

Rebooting...

**************************************************************************************************

FBester1976, 2024-03-17:
Never seen those errors before, try VSCode

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md

RS33D, 2024-03-21:
Hello 

 

Have tried VScode and have mangaed to get it to work (sounds etc), but after fiddling around with the sounds, changing it from the orginal to the caterpillar truck sound it doesn’t work, I also played with channel assignments

So I uploaded the orginal code (as per the one that works) and it still doesn’t work

 

Please see the below google drive video for what happens when I power it on 

https://drive.google.com/file/d/1pOtpVrpAXErzeCzaloz0xr0UTYyJBJzV/view?usp=sharing

 

when plugged in to the serial monitor (esc and Y cable still on) it gives me this (but early in my testing I rember seeing something about Ch1 being an issue but I cant seem to make that come up again, might have been about an offset, not to sure though) 

 

 

--- Terminal on COM3 | 115200 8-N-1 

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time 

--- More details at  <https://bit.ly/pio-monitor-filters> https://bit.ly/pio-monitor-filters 

--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

, check transmitter & receiver! 

[... 182 lines cut ...]
rst:ets Jul 29 2019 12:21:46 

rst:ets Jul 29 2019 12:21:46 

rst:ets Jul 29 2019 12:21:46 

rst:ets Jul 29 2019 12:21:46 

rst:ets Jul 29 2019 12:21:0x0

 

(DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 

(SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 (DOWNL0x6 (SDIO_RESET),boot:0x0 

(DOWNL0x6 (SDIO_RESET),bootSDIO_FEI_FEO_V2)) 

waiting for download 

 

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32 

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md

FBester1976, 2024-03-22:
Hi, not sure what all you changed. Have not seeb those errors before.

https://drive.google.com/file/d/1pOtpVrpAXErzeCzaloz0xr0UTYyJBJzV/view?usp=sharing

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md

RS33D, 2024-03-22:
Hello, have managed to get it to work reliably now, just need to sort out
my channel assignments, unsure of how to turn headlights on and off, adjust
high or low beam and turn on side lights

FBester1976, 2024-03-22:
https://youtube.com/playlist?list=PLhpVbuijh28IvC5ml1K5NiqUKX2A34Fp1&si=OuBoI1wKSks4r1Aq

https://youtu.be/TLT7ZF_BZVU?si=bYFdPg2GUVyOH7T2

RS33D, 2024-03-24:
Awesome, thanks for that, have managed to get it all working 

 

Thanks so much for your help, much appreciated 

 

https://youtube.com/playlist?list=PLhpVbuijh28IvC5ml1K5NiqUKX2A34Fp1&si=OuBoI1wKSks4r1Aq 

https://youtu.be/TLT7ZF_BZVU?si=bYFdPg2GUVyOH7T2