TheDIYGuy999/Rc_Engine_Sound_ESP32 #70: Brownout detector was triggered

bereldhuin, 2024-03-17:
Hi,

I'm currently testing the sound module on a breadboard (sound + headlight and tail light only).

Everything is working fine until I raise the volume, at a not so high level.
At this time, the ESP32 reboots with a message "Brownout detector was triggered".

After looking for solutions on Internet, I tried using a better power supply (a Raspberry official 3A one) directly on 5V pin, or using a 47uF capacitor between GND and 3.3V, I didn't manage to solve this problem.

Do you have any idea ? 
I saw that using a bredboard with PAM8403 is not ideal. Will a real PCB solve this issue ?

rengels, 2024-03-17:
The power regulators might not be fast enough if there is a sudden loud sound.
You might want to use different ones for the ESP and the audio amplifier, so that at least the esp32 supply is stable.

An oscilloscope might help in checking the voltages. There you can also see if your amplifier is creating noise on the power supply side.

I don't think that a PCB will solve the issue unless your cables are too thin or you'r routed your cables in an extremely bad way. It might be a loudspeaker with a too low resistance.

bereldhuin, 2024-03-17:
Thanks for your answer Ralph.

I'm using standard 4ohms, 5W speakers :
https://www.amazon.fr/dp/B08RY2VHZG?psc=1&ref=ppx_yo2ov_dt_b_product_details

From what I understood (and re-reading the scheamtics), I should use a MP1584 for AMP, and the BEC power for ESP32, right ?

rengels, 2024-03-18:
The AMP and voltage converter is pretty flexible. If you have a 5V battery, you might get away without any regulator.
Same for the AMP. If you want to use 12v for audio you can pick something else.

But the core of your issue seems to be that the voltage drops.
You should check that the VCC is stable enough.

bereldhuin, 2024-03-19:
I separated the power supply of the ESP32 (powered by microusb) and the AMP (powered by an external usb adapter) and everything is working fine now.

Even the interferences I got when the transmitter and receiver were on are gone. :-) 

Thanks again for your help.

closed 2024-03-19 by bereldhuin