TheDIYGuy999/Rc_Engine_Sound_ESP32 #71: Microprocessor Minimum Requirements

sk8board, 2024-03-17:
I noticed the documentation for this project does not list a minimum requirement for the micro. Would you know the minimum processor speed, RAM, ROM, and flash size for this project?

For example, will any version of ESP8266 or ESP32 or Pi Pico work?

https://www.espboards.dev/blog/esp32-soc-options/#esp32-versions-comparison

https://www.waveshare.com/wiki/RP2040-Zero#Specification

rengels, 2024-03-18:
The SW uses the dual core of the ESP32 and both cores are fairly maxed out.
Regarding the Flash I think we are at 60% usage.
The only viable option without many SW changes might be ESP32S3.

The SW isn't optimized at all, so you might be able to change it. But would need to invest a lot of time.

If I were you, I would just just the same HW that everyone else uses.

rengels, 2024-03-19:
I understand your problem now.

While playing audio does not need a lot of processing power, the Rc_Engine project does a lot of computation and mixing of sounds.
That pretty much uses up the 200MHz core at higher RPMs (the interrupt get's triggered more often at higher RPMs).

I am sure you can create a small, stripped down version of an engine sound generator on a much smaller CPU.
You would need a digital-to-analog converter though.

The Raspberry is powerfull enough for this, there is however one big downside.
You will need to implement this yourself.

So this discussion is pretty much theoretical, unless you manage to set up a project on your own. The code of the Rc_Engine_Sound project is not portable at all.

sk8board, 2024-03-19:
Thank you for your response.

Based on your input, I decided to look for the smallest ESP32 and found the [TinyPico Nano](https://www.tinypico.com/tinypico-nano) has the same specs as the ESP32-WROOM that you used. I designed a preliminary PCB to determine package size which includes an Express LRS (ELRS) receiver, 3A DC-DC power supply, Brushed ESC, and LED driver (58x38mm). The audio amplifier would need to be a separate board due to limited space on smaller WPL models. Changing to surface mound components would reduce size even more.

https://oshwlab.com/sk8board/esp32-elrs-receiver

I am looking at my options and would like to understand the challenges of adding Express LRS (ELRS) to your project. I was surprised to see some of your code is specific to FlySky transmitters. Is your code deeply connected to a FlySky transmitter? Would it be painfully difficult to add an ELRS transmitter and receiver to your code?

rengels, 2024-03-19:
Hi,
great that you found a nice HW.
The code is not specific to any transmitter. You can add your own stuff.

However, let me just repeat a part of my last message:

> So this discussion is pretty much theoretical, unless you manage to set up a project on your own. The code of the Rc_Engine_Sound project is not portable at all.

You would need someone with good experience in C and embedded programming.
I would also estimate the necessary time to port the code to another architecture and rip out the unnecessary parts at 80 hours.

It's probably easier to start from scratch and just copy the audio conversion tools and some of the samples.