TheDIYGuy999/Rc_Engine_Sound_ESP32 #74: Through Hole Vs SMD

SteveGoral, 2024-06-09:
Good Afternoon,

Looking at PCBway, getting one of these boards assembled isnt very cost effective.  It's comes out at $88, and there are still components to source even after they have mounted the SMD parts.  Is there any chance of you releasing the files for the Through Hole board so I can get tat produced and then solder the whole thing myself?

TheDIYGuy999, 2024-06-09:
See hardware folder

SteveGoral, 2024-06-09:
Amazing thanks for the reply.  Is the functionality the same?  I've just retried a quote on PCBway and got a much more wallet friendly price, so just mulling it over.

fiechr, 2024-06-11:
As far as I remember, I got all the parts from AliExpress for maybe 20-30 EUR and 5-10 empty PCBs for less then 10 EUR. Don't pin me down on the exact prices, but it was definitely a lot less than $88 for multiple fully built controller boards.

I can't imagine through hole to be more cost effective, neither for production nor for sourcing parts (apart from parts that are through hole only, of course). If you don't mind soldering, I would always recommend to do it yourself.

I also recommend using the PCB with the switch mode voltage regulator module.

For more help you may want to check out this forum's thread (maybe you already did):
https://www.rc-modellbau-portal.de/index.php?threads/esp32-arduino-rc-sound-und-licht-controller.7183/