TheDIYGuy999/Rc_Engine_Sound_ESP32 #76: What is the license

rengels, 2024-06-17:
I just noticed: there is no information about the License.

I think there should be some License.txt file somewhere. Possibly MIT or GPL. Maybe even LGPL (because, why not).
Right now, it is a little bit fuzzy, if we are even allowed to do any changes.

JohnEdwa, 2024-08-31:
[Without a license file the owner has full copyright of the code - all rights reserved ](https://choosealicense.com/no-permission/) - and anyone who wants to use it in any way has to get explicit permission from the creator.

rengels, 2024-09-01:
Of course you are right.
But that would mean that I am running legal risks by just compiling the source code and flashing it on my esp32.

I think you understand my issue. We need a clearly stated license, possibly an open source license like MIT or GPL.