TheDIYGuy999/Rc_Engine_Sound_ESP32 #77: v9.12 error: 'timerAlarmWrite' was not declared in this scope;

Bananasmoothie1975, 2024-08-19:
On compiling v9.12 (actually now all versions I've tried), error of:

> error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?

for all instances of 'timerAlarmwrite'.

Using IDE v2.3.2, ESP32Devkit1 board.
Have also had some issues with not finding folders, particularly "soc" folder which i can't find anywhere using File Exploroer either.
Thanks

Bananasmoothie1975, 2024-08-21:
Looks like it's something to do with this (scroll to the Timer section):

https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#compilation-flags

Do I have to replace ALL the different API's or is there some sort of code auto update function?

TheDIYGuy999, 2024-08-21:
Use VS Code with PlatformIO instead of Arduino

Bananasmoothie1975, 2024-08-21:
> Use VS Code with PlatformIO instead of Arduino

Firstly, thanks for you reply and for all the effort in your RC controller, I have enjoyed all the videos and years ago built a PWM version for my HG P407 RC (modded to suit my own requirements) :D

I've now worked my way to a compileable sketch, I had to install the previous Espressif ESP32 API (v2.xxxx, down from v3.03) so the timer functions etc were correct.   

I did download the VS Code software, but I have no idea what it does or how to use it :s 

I'm now at the point of trying to get the SBUS working through my Flysky i6X and iA10B Rx, the serial monitor isn't showing any channel info, just EEPROM and core reg dump stuff but none of the individual channel stuff even shows up (none of the channel serialprint info).
SBUS to pin36 ("VP") if that helps.

FYI, I'm trained in mechanical systems, not electronics/code, but learning ;)

TheDIYGuy999, 2024-08-21:
This means it's compiling but crashing anyway. You need to use the correct, even older Espressif API. You should watch some VS Code tutorials and you will see it's way easier than Arduino, because it's installing all the correct API's and libraries automatically for you.

Bananasmoothie1975, 2024-10-15:
Eventually got VSCode and platformio etc working, code now compiles and uploads, thanks.

closed 2024-10-15 by Bananasmoothie1975