TheDIYGuy999/Rc_Engine_Sound_ESP32 #78: problem with L120 profile

mroversi, 2024-09-07:
[image]
[image]
[image]

Hi,  is the code that is not working or I'm doing some errors?
I also made a new try with different  .ini
[image]

Would be nice to have the profile on the web app too
thanks

rengels, 2024-09-08:
Hi,
nice screenshots.

It would be better to push your changes, or at least post the diff here (instead of a screenshot).
From the error message I think that something with your environment is not OK.

Did you build prior to making the changes? As a baseline?

mroversi, 2024-09-09:
I just uploaded it ad didn't made any changes , just to upload the sw on the esp32.