TheDIYGuy999/Rc_Engine_Sound_ESP32 #81: shaker only work with ps2 gamepad motor

Peyutron, 2024-11-21:
Hi!! Very good and excellent work. I have a slightly strange problem with the shaker. I have used a motor from a PS2 controller and it works very well. The problem is that if I try to connect a different type of motor like the one used by the smoke generator blower (just the motor) it doesn't do anything... 
Also if I try a different type of engine. I have tested the breadboard with a transistor of the same type and both work, both the one I have for the shaker and the one for the blower.
I only have to ask if when defining the object shakerMotor.begin(SHAKER_MOTOR_PIN, 13, 20000); // Timer 13, 20kHz

Can the frequency make these motors not work?
I have tried different values ​​like 0, 100, 1000, 10000 and 50000.

Has this happened to anyone too?

Greetings!!!

TheDIYGuy999, 2024-11-22:
Thanks, adjust it here:
[image]

Peyutron, 2024-11-24:
Hi! Yes, I set all those values ​​to 255. Even so, the blower didn't work.
The problem was the resistor with the transistor (I'm using a BC547) which was 1K, I changed it to 330 and it works!!!

Now I can adjust the values.

Thank you very much and great job!!