TheDIYGuy999/Rc_Engine_Sound_ESP32 #82: Hello everyone. i am having issues with upload the code of the older version 6.4. i have limited knowledge in coding and request someone to help me out please it will mean alot

Ritz-888, 2024-12-24:


TheDIYGuy999, 2024-12-24:
Why using an older version? If you have limited knowledge, you should try this first: https://thediyguy999.github.io/TheDIYGuy999_ESP32_Web_Flasher/flash.html

Ritz-888, 2024-12-24:
> Why using an older version? If you have limited knowledge, you should try this first: https://thediyguy999.github.io/TheDIYGuy999_ESP32_Web_Flasher/flash.html

im using the older version since i only want the sound and lights system not the dashboard. i have made my own rc trailer truck and its interior is not that good so iv decided not to have the displays

TheDIYGuy999, 2024-12-24:
You can use the current version, even if you don't have the dashboard. Just disable it.

Ritz-888, 2024-12-24:
> You can use the current version, even if you don't have the dashboard. Just disable it.
i tried the new version but i am finding it complicated. i am in year 11 and have very little knowledge about codding. i will really appreciate if you can help me step by step.

Ritz-888, 2024-12-24:
> You can use the current version, even if you don't have the dashboard. Just disable it.

i tried the new version but i am finding it challenging. i am in year 11 and have very little knowledge in codding. so if possible can you help me step by stem so i could complete my truck. i will be so thankful for your assistance.

TheDIYGuy999, 2024-12-24:
Have a look here: https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32

here: https://www.facebook.com/groups/969278734997783

and here: https://www.rc-modellbau-portal.de/index.php?threads/esp32-arduino-rc-sound-und-licht-controller.7183/

Ritz-888, 2024-12-24:
[image]
How can i solve this errors