TheDIYGuy999/Rc_Engine_Sound_ESP32 #86: please help  CH1: signal out of auto calibration range, check transmitter & receiver!

xskrtzx, 2025-02-12:
Hi There

I ran into a lot of dead ends because of my abilities. 

First they said the battery voltage was low, where I had installed the regulator according to the schematic and components required. then I turned off the battery protection system,

it continues with 
 **CH1: signal out of auto calibration range, check transmitter & receiver!**
what should I set? I didn't change anything programmed, sorry I'm new to RC and the ins and outs of it. 

I use **RC RGT EX86190**
which uses the **dumborc X8e-150 gs07 remote**
and **X6DC receivers**. I just connected the **esp32** to the **sound amp**, then the **receiver** and **regulator**, all complete with the necessary **resistors** but the gas pedal didn't respond soundly when pressed

which parts do I need to setup in the program? I just want to get the sound out. sometimes the speaker just makes noise, I use **pam8403**

I hope you can help, I really appreciate it.

FBester1976, 2025-02-12:
Hello,

Which version of the code are you using?

I assume you are using PWM? I have seen on the forum that you must use code version 9.11.0 or older, on the newer versions the PWM does not work. I have not tested this, but people that struggled used the older code version and came right.

[https://www.rc-modellbau-portal.de/index.php?threads/esp32-arduino-rc-sound-und-licht-controller.7183/post-580170](url)
[https://www.rc-modellbau-portal.de/index.php?threads/esp32-arduino-rc-sound-und-licht-controller.7183/post-580252](url)

On the 2_Remote.h you must also make sure you select the correct remote, and have the SBUs and IBUS commented out (insert // before code)

[image]

You must also have a cable between your receiver and sound card for the 3 channels defined in the code. Please take note of the receiver channel and where the cable must go on the sound card.

[image]

Also not sure if you want to use the gearbox? Then you need to adjust the code to tell the SW which receiver channel will give the signal. I think on the DumboRC Ch4 is a 3 position switch.

[image].

I hope this helps...

Regards,

Frikkie

xskrtzx, 2025-02-13:
> Hi,
> 
> Glad you got the code sorted.
> 
> Can you send a picture of your setup? Do you 

I am not sure you can see clearly. Because they are still on breadboard. I also make my own diy soldering board but still process. I will finish it after this whole tested correctly.

I also supply my self with spare Ic, capacitor, resistor and esp32. Just for compare and try and error. So far I got one esp32 breakdown (it won't able to upload.

Well here the picture, thankyou for you time and effort for helping me out
Basically my wiring diagram is same as diyguy999 tutorial. But if you say pam8403 can operate 2.5-5v so now what the problem. With the same wiring way
- when I use my rc batteries to regulator (sound won't came out
- but when I use USB C direct to esp32 is connect. Maybe trouble in esp32 VIN?
- 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

FBester1976, 2025-02-13:
It is impossible for me to check the breadboard connections.

Please note there is 2 different 5V inputs, one from the ESC that is connected to the headers, and one from the 5V regulator.

[image]

[image]

So with the printed PCB I know it is important to either have the ESC plugged into the headers to supply 5V, or have a jumper in place. 

Did you change the code in the Remote tab to have None for Ch3. I see no wire from the RX to your breadboard. Not sure if it will work correctly if it gets no signal for Ch3.

Not sure where in the world you are, but ordering the PCB's from PCBWay to South Africa was not that expensive, and makes live a lot easier!

xskrtzx, 2025-02-28:
> It is impossible for me to check the breadboard connections.

Hi, 

Finally I can finished this stuff after So much drama. Lol

So I continued with my DIY PCB. I'm just lazy to sit still and wait for the pcb order. All the components are complete so why wait. On your advice I use pwm. And version 9.11 For rc RGT ex86190. But I just confused how to activated horn. I just try all my button and set in the remote adjustment but still not working

[image]

closed 2025-03-02 by xskrtzx