TheDIYGuy999/Rc_Engine_Sound_ESP32 #92: no sound

radu022003, 2025-07-03:
Hello,
my hardware at the moment looks like the prototype on the documentation page.
Instead of the RC receiver, I am using a servo tester.
I have commented out all communication protocols in order to use the classic PWM RC signals (i hope that this is the correct way).

I have connected the output of the servo tester through 330ohm resistors to pins D13 and D14.
I have sound output because I get the "fuel low voice message"

when the device starts I can hear 3 "clicks", but no sound anymore. I have no lights connected. My purpose is to use it on a RC boat.

the serial monitor is writing this:
**************************************************************************************************
TheDIYGuy999 RC engine sound & light controller for ESP32 software version 9.13.0
https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32
Please read carefully: https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md
XTAL Frequency: 40 MHz, CPU Clock: 240 MHz, APB Bus Clock: 80000000 Hz
Internal RAM size: 382096 Byte, Free: 356096 Byte
WiFi MAC address: E8:6B:EA:D4:8F:E8
Core 0 reset reason: 1: POWERON_RESET
Core 1 reset reason: 14: EXT_CPU_RESET
**************************************************************************************************

EEPROM read.
current eeprom_id: 5
change it for default value upload!

Battery protection calibration data: ----
RESISTOR_TO_BATTTERY_PLUS: 9400 Ω
RESISTOR_TO_GND: 1000 Ω
DIODE_DROP: 0.31 V

ESC calibration data: ----
ESC pulse span: 600 (Used to adjust the top speed: 500 = full ESC power, 1000 = half ESC power etc.)
ESC takeoff punch: 0 (Usually 0. Enlarge it up to about 150, if your motor is too weak around neutral.)
ESC reverse plus: 0 (Usually 0. Enlarge it up to about 220, if your reverse speed is too slow.)
ESC ramp time for crawler mode: 10 (about 10 - 15), less = more direct control = less virtual inertia)
**************************************************************************************************

ENABLE_WIRELESS option disabled, no WiFi configuration or ESP-Now Trailer!
-------------------------------------
Battery voltage: 3.79 V
Cutoff voltage per cell: 3.30 V
Fully charged voltage per cell: 4.20 V
Number of cells: 3 (3S battery detected) Based on setpoint: 8.55 V
Battery cutoff voltage: 3.30 V (3 * 3.30 V)
-------------------------------------
... PWM communication mode active.
-------------------------------------
Standard ESC mode configured. Connect crawler ESC to ESC header. RZ7886 motor driver not usable!
-------------------------------------

TheDIYGuy999, 2025-07-03:
Change this:

[image]

radu022003, 2025-07-03:
Thank you for the fast response, I did the change accordingly, now the 3 clicks are gone, it is complete silence.

```
**************************************************************************************************
TheDIYGuy999 RC engine sound & light controller for ESP32 software version 9.13.0
https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32
Please read carefully: https://github.com/TheDIYGuy999/Rc_Engine_Sound_ESP32/blob/master/README.md
XTAL Frequency: 40 MHz, CPU Clock: 240 MHz, APB Bus Clock: 80000000 Hz
Internal RAM size: 382844 Byte, Free: 356844 Byte
WiFi MAC address: E8:6B:EA:D4:8F:E8
Core 0 reset reason: 1: POWERON_RESET
Core 1 reset reason: 14: EXT_CPU_RESET
**************************************************************************************************

EEPROM read.
current eeprom_id: 5
change it for default value upload!

ESC calibration data: ----
ESC pulse span: 600 (Used to adjust the top speed: 500 = full ESC power, 1000 = half ESC power etc.)
ESC takeoff punch: 0 (Usually 0. Enlarge it up to about 150, if your motor is too weak around neutral.)
ESC reverse plus: 0 (Usually 0. Enlarge it up to about 220, if your reverse speed is too slow.)
ESC ramp time for crawler mode: 10 (about 10 - 15), less = more direct control = less virtual inertia)
**************************************************************************************************

ENABLE_WIRELESS option disabled, no WiFi configuration or ESP-Now Trailer!
-------------------------------------
Warning, BATTERY_PROTECTION disabled! ESC with low discharge protection required!
-------------------------------------
... PWM communication mode active.
-------------------------------------
Standard ESC mode configured. Connect crawler ESC to ESC header. RZ7886 motor driver not usable!
-------------------------------------

Transmitter channel offsets (calculated, if channelAutoZero[] = true):
 CH1: -25 µs
 CH3: -26 µs
 CH5: signal out of auto calibration range, check transmitter & receiver!
```

radu022003, 2025-07-04:
i have looked into the line 2187 of the file src.ino, by placing a serial.print there for the channel3. I could observe that the readings are made and the value changes with the potentiometer position, of the servotester. but the serial is printing only one time, at the beginning, then nothing more.

closed 2025-07-18 by radu022003